paulvangentcom/heartrate_analysis_python #2: Sampling Frequency of 44100 Hz ValueError: min() arg is an empty sequence

SreehariRamMohan, 2017-11-26:
Hi

I am running into an issue with using this module to calculate the BPM of a .wav file sampled at 44100Hz (Python 3.6)

```
This is the error which I get: 
 Traceback (most recent call last):
  File "/Users/JohnDoe/Desktop/TensorFlow_Programs/Heart-Sounds-Deep-Learning/heart-rate-analysis-module/UsingHeartbeatModule.py", line 15, in <module>
    measures = hb.process(data, 44100)
  File "/Users/JohnDoe/Desktop/TensorFlow_Programs/Heart-Sounds-Deep-Learning/heart-rate-analysis-module/heartbeat.py", line 205, in process
    fit_peaks(hrdata, rol_mean, fs)
  File "/Users/JohnDoe/Desktop/TensorFlow_Programs/Heart-Sounds-Deep-Learning/heart-rate-analysis-module/heartbeat.py", line 124, in fit_peaks
    working_data['best'] = min(valid_ma, key = lambda t: t[0])[1]
ValueError: min() arg is an empty sequence
```

I am using your heart-rate module to calculate the BPM (shown below):

```
import heartbeat as hb
from scipy.io import wavfile as wav
import numpy
import pandas as pd

RAW_CSV_NAME = "/Users/JohnDoe/Desktop/TensorFlow_Programs/Heart-Sounds-Deep-Learning/pls.csv"
fs, data = wav.read('/Users/JohnDoe/Desktop/output_test.wav')
df = pd.DataFrame(data)
df.columns=['hart']
df.to_csv(RAW_CSV_NAME, index=False, header="hart" )

print("Sampling frequency " + str(fs)) --> prints 44100
hrdata = hb.get_data(RAW_CSV_NAME, column_name = 'hart')
measures = hb.process(data, 44100)

print(measures['bpm']) #returns BPM value
print(measures['lf/hf']) # returns LF:HF ratio

#Alternatively, use dictionary stored in module:
print(hb.measures['bpm']) #returns BPM value
print(hb.measures['lf/hf']) # returns LF:HF ratio
```

Here is the code for recording with PyAudio
```
import pyaudio
import wave

CHUNK = 1024
FORMAT = pyaudio.paInt16
CHANNELS = 1
RATE = 44100
RECORD_SECONDS = 30
WAVE_OUTPUT_FILENAME = "/Users/sreeharirammohan/Desktop/output_test.wav"

p = pyaudio.PyAudio()

stream = p.open(format=FORMAT,
                channels=CHANNELS,
                rate=RATE,
                input=True,
                frames_per_buffer=CHUNK)

print("* recording")

frames = []

for i in range(0, int(RATE / CHUNK * RECORD_SECONDS)):
    data = stream.read(CHUNK)
    frames.append(data)

print("* done recording")

stream.stop_stream()
stream.close()
p.terminate()

wf = wave.open(WAVE_OUTPUT_FILENAME, 'wb')
wf.setnchannels(CHANNELS)
wf.setsampwidth(p.get_sample_size(FORMAT))
wf.setframerate(RATE)
wf.writeframes(b''.join(frames))
wf.close()

```

I was wondering if you could help me fix this Sampling Rate Issue. Any Guidance would be greatly appreciated!

paulvangentcom, 2017-11-27:
Hi SreehariRamMohan, 

What's happening is that, when trying various fits for the moving average, none of them contain correctly detected peaks. The algorithm is not designed with audio signals in mind, so there might be an issue there. 

Could you share your audio recording with me so I can have a look at the raw data?

-Paul

SreehariRamMohan, 2017-11-27:
Sure I have attached a link to the drive .wav file of my heartbeat
https://drive.google.com/file/d/1NmhJuIKtXNpTNyBtc2sv9Rz5dvZDB3LP/view?usp=sharing

SreehariRamMohan, 2017-11-27:
I convert the raw .wav file to a .csv using wav.read()
```
fs, data = wav.read('/Users/JohnDoe/Desktop/output_test.wav')
df = pd.DataFrame(data)
df.columns=['hart']
df.to_csv(RAW_CSV_NAME, index=False, header="hart" )
```

paulvangentcom, 2017-11-28:
Hi Sreehari,

The main issue is that the algorithm is designed with electrical or optical (PPG) sensors in mind. In this data the cardiac cycle is only expressed as S1 (first heart sound, the first thump in your audio). However in the audio S2 (second heart sound) in the diastole phase is also recorded.

The error correction right now hinges on rejecting illogical beat-beat intervals given the surrounding signals, and because the amplitude of S1 and S2 are very close together as shown on the image below, this creates problems.

[image]

I'll take a closer look over the weekend and see if I can implement a rudimentary 'audio detection mode' that may help you.

SreehariRamMohan, 2017-11-29:
Ok thank you for trying!

paulvangentcom, 2017-12-04:
Give me a few more days, unfortunately with the holidays last weekend there was no time to look at this :).

SreehariRamMohan, 2017-12-13:
@paulvangentcom  Sorry to bother you, just wondering if you have made any progress so far :)

paulvangentcom, 2017-12-14:
Hi Sreehari,

Apologies for the delay, I broke my thumb earlier so I've not had much coding time available.

I've pushed an experimental function to the repo (in the folder 'experimental'). Please be aware that the peak detection is not super accurate yet. Since the variability measures are noise-sensitive they should **not** be trusted. The BPM value as you requested, is much more noise-resistant and should be a pretty good estimate of the real BPM in the audio file, given there is not too much noise..

Usage is similar to the main module:

`import heartbeat_audio_experimental as hb_audio

measures = hb_audio.process('heartbeat.wav')

print(measures['bpm'])

hb_audio.plotter()
`

This gives:

[image]

SreehariRamMohan, 2017-12-16:
Hi, 
Thanks for all of your hard work!
I ran this script and it worked perfectly on my Macbook Pro.
However when I tried to run it on my raspberry pi 3 I ran into an error:
```
resampling signal to 1000Hz
detecting peaks
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "Project_GUI.py", line 307, in record
    self.get_BPM_thread = threading.Thread(target=self.getBPM(), daemon=True)
  File "Project_GUI.py", line 410, in getBPM
    measures = hb_audio.process(self.WAVE_OUTPUT_FILENAME)
  File "/home/pi/Desktop/Heart-Sounds-Deep-Learning/heart-rate-analysis-module/experimental/heartbeat_audio_experimental.py", line 89, in process
    rol_mean = hb.rolmean(hb.working_data['hr'], 0.75, new_samplerate)
  File "/home/pi/Desktop/Heart-Sounds-Deep-Learning/heart-rate-analysis-module/heartbeat.py", line 112, in rolmean
    rol_mean = np.mean(rollwindow(data_arr, int(windowsize*sample_rate)), axis=1)
  File "/home/pi/Desktop/Heart-Sounds-Deep-Learning/heart-rate-analysis-module/heartbeat.py", line 100, in rollwindow
    return np.lib.stride_tricks.as_strided(data, shape=shape, strides=strides)
  File "/home/pi/.local/lib/python3.5/site-packages/numpy/lib/stride_tricks.py", line 102, in as_strided
    array = np.asarray(DummyArray(interface, base=x))
  File "/home/pi/.local/lib/python3.5/site-packages/numpy/core/numeric.py", line 531, in asarray
    return array(a, dtype, copy=False, order=order)
ValueError: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.

```

At first I thought this was an error with the Raspberry Pi 3 not having enough RAM, so I created 26 gigabytes of swap space on the Pi however this did not fix the issue. 

Do you have any ideas as to why this issue is happening?

paulvangentcom, 2017-12-16:
My guess would be you are using 32-bit python on the pi. Could you open the interpreter and check this?

paulvangentcom, 2017-12-16:
Please clone the most recent commit. I've changed the sampling rate calculation method. Let me know if this fixes it for you. It should work now. If it doesn't, try changing '1000' at line 81 to '100' so you resample to 100Hz in stead of 1000Hz.

SreehariRamMohan, 2017-12-20:
@paulvangentcom 

Sorry for late response, I've had a high fever for the last few days so I wasn't able to get the recordings.

I've attached 5 30 second heartbeat sounds. One thing I found strange was that these recordings had more static than the previous ones. I was wondering if you could maybe filter the sound before taking the BPM?

heartbeat #1
https://drive.google.com/file/d/1wpSxodLQtFwSoY5vWB41rmyGp1FxUG7e/view?usp=sharing

heartbeat #2
https://drive.google.com/file/d/1XkhtKpBCq71mjLvJ5WsqQjmyb49uLSPV/view?usp=sharing

heartbeat #3 
https://drive.google.com/file/d/1gKuvgfwf9MLriGKvi9TikDudSw3_ru8u/view?usp=sharing

heartbeat #4
https://drive.google.com/file/d/1QhyRPvV72oEIXEtBUgsWNuMkEhW29uuF/view?usp=sharing

heartbeat #5
https://drive.google.com/file/d/14sUSfIO3rQAb45ipyq_fmD-zn15pN2vL/view?usp=sharing

Let me know if you need more heartbeat recordings. 

Looking forward to seeing the improvements!

SreehariRamMohan, 2018-01-02:
@paulvangentcom  By the way the commit you made on December 16, 2017 fixed the memory issue on my pi. Tell me if you have any further questions.

paulvangentcom, 2018-01-02:
Thanks for that. When there's some time I'll try fine-tuning the audio detection.

- Paul

closed 2018-01-02 by paulvangentcom