paulvangentcom/heartrate_analysis_python #4: only integer arrays error

ysaidcan, 2018-02-09:
When i give the input HR data to the library, i got an error:
peaksy = hrdata[np.where((hrdata > rol_mean))[0]]

TypeError: only integer arrays with one element can be converted to an index

My code is:

import bitalino
import numpy as np
import time
import matplotlib.pyplot as plt
import peakdetect
import heartbeat as hb

# Mac OS
# macAddress = "/dev/tty.BITalino-XX-XX-DevB"

# Windows
#macAddress = "20:16:07:18:13:64"
macAddress= "20:17:09:18:46:90"
   
#device = bitalino.BITalino(macAddress)
time.sleep(1)

srate = 1000
nframes = 100
threshold = 5

data=np.loadtxt('baselineHr.txt')

ECG=data[:, -1]

ECGmean=np.mean(ECG)
        

peaks = peakdetect.peakdetect(np.array(EDA), lookahead=2,delta=threshold)
indexes = []
for posOrNegPeaks in peaks:
    for peak in posOrNegPeaks:
        indexes.append(peak[0])
                
peaksECG = peakdetect.peakdetect(np.array(ECG),delta=ECGmean/20)
indexesECG = []
for posOrNegPeaks in peaks:
    for peak in posOrNegPeaks:
        indexesECG.append(peak[0])
        
ECG_data_arr=[]

for i in range(0,len(ECG)):
    ECG_data_arr.append(np.int(ECG[i]))
            

sp=np.fft.fft(ECG_data_arr)

freq = np.fft.fftfreq(len(ECG_data_arr), d=1/srate)

data = ECG_data_arr
[ECG_data.xlsx](https://github.com/paulvangentcom/heartrate_analysis_python/files/1711699/ECG_data.xlsx)

measures=hb.process(data,1000)

paulvangentcom, 2018-02-10:
Hi Ysaidcan,

From what I can see three things are happening. 

1. Your data sampling speed seems to be around 100.0 Hz and not 1000
2. The type-error can be fixed by re-casting the data as np array.
3. You're using ECG-type measuring devices. This is fine, but allow me to explain:

The algorithm was originally designed for PPG-type sensors. ECG-type sensors give very narrow peaks and different noise characteristics. The algorithm can still handle them, although you either need to up your sampling speed, or resample the signal:

```python
import heartbeat as hb
from scipy.signal import resample

data = np.asarray(ECG_data_arr, dtype=np.float32)
resampled = resample(data, len(data)*10)

#Now fs it is 1000Hz, since I upsampled it * 10
measures = hb.process(resampled, 1000.0, calc_fft=False) 
```

**Please note** that I turned the frequency-domain measures off (`calc_fft=False`). After upsampling, the fft method currently upsamples again which causes performance issues. I'll try to push a fix for this later today.

The above code, using your data, for me gives:
[image]

You can also upsample and plot the RR-intervals to estimate your breathing pattern:
[image]

I'll push the breathing visualisation with the update today or tomorrow.

- Paul

paulvangentcom, 2018-02-12:
I pushed the fft performance fix last night.  I'm assuming this and the above fixed your issue. Let me know otherwise.

- Paul

closed 2018-02-26 by paulvangentcom