paulvangentcom/heartrate_analysis_python #5: Crash on noicy signal

yurasmol, 2019-01-19:
In a noicy signal in fit_peaks function if calculated _bpm always upper than bpmmax (180 bpm), then valid_ma is empty list and I have error "min() arg is an empty sequence"

paulvangentcom, 2019-01-20:
Hi Yurasmol,

What sampling rate is the attached file recorded at? It's so noisy that I'm having issues seeing where the heart rate signal is.

[image]

Let me know and I'll see what I can do with some filtering.

Cheers,
Paul

paulvangentcom, 2019-01-28:
Hi Yurasmol,

Please pull the latest version. I've introduced custom exception handling to remedy your issue and make the error more verbose. These situations now trigger a UserWarning `BadSignalWarning`.

Note that you still need to catch any warnings and exceptions if you want execution to continue. Consider an example using 'data.csv' from the repo (sampled at 100Hz):

```python
import heartpy as hp

if __name__ == '__main__':
    #Let's simulate empty valid_ma during iteration flow example 
    #by setting bpmmax to 40 (real is 58.92)
    bpmmax = [60, 40, 130] 
    data = hp.get_data('data.csv')
    for i in bpmmax:
        print(i)
        try:
            wd, measures = hp.process(data, 100.0, bpmmax=i)
        except hp.exceptions.BadSignalWarning as w:
            print(w) #display message
            pass #or just pass over it if you want
```

Note the above example steps over the UserWarning `BadSignalWarning` but will fail on any other exception. You'd need to catch that separately.

Lastly the non-inheritance of bpmmin and bpmmax has been fixed. Thanks for catching that!

I'll have a look at the other issue later in the week as I'm on holidays now so my time is limited.

Cheers,
Paul