paulvangentcom/heartrate_analysis_python #6: Threshold algorythm

yurasmol, 2019-01-19:
If the first stage of signal have the distortions, the second (good) stage fully 
[samp.pdf](https://github.com/paulvangentcom/heartrate_analysis_python/files/2775786/samp.pdf)
marked as rejected

paulvangentcom, 2019-01-20:
Hi Yurasmol,

Interesting. Can you provide the raw data files for me? I'll have a look.

Cheers,
Paul

yurasmol, 2019-01-21:
Hi Paul!

This is my RAW signal (sample rate is 100 Hz) and the test module where I four times cut 1/3 starting portion (noisy) of the signal.
I think that your threshold algorythm get information from the start portion of a signal (noisy in my case) and therefore rest of signal marked rejected.

Yuri

  ----- Original Message ----- 
  From: Paul van Gent 
  To: paulvangentcom/heartrate_analysis_python 
  Cc: yurasmol ; Author 
  Sent: Sunday, January 20, 2019 5:11 PM
  Subject: Re: [paulvangentcom/heartrate_analysis_python] Threshold algorythm (#6)

  Hi Yurasmol,

  Interesting. Can you provide the raw data files for me? I'll have a look.

  Cheers,
  Paul

  —

from matplotlib.backends.backend_pdf import PdfPages
import heartpy as hp
import matplotlib.pyplot as plt
import numpy as np
import warnings, sys, os
from scipy.signal import savgol_filter

pdf = PdfPages("test.pdf")	# Create the PDF document

data = hp.get_data('RAW.txt')

f = plt.figure(0, figsize=(12, 6))
plt.plot(data, linewidth=0.2)
pdf.savefig(f, bbox_inches="tight")

#f  = plt.figure(1, figsize=(12, 6))
#working_data, measures = hp.process(data, 100, windowsize=1, freq_method='welch', calc_freq=True)
#p  = hp.plotter(working_data, measures, show=False)
#pdf.savefig(f, bbox_inches="tight")

data = data - savgol_filter(data, 501, 4)

f = plt.figure(2, figsize=(12, 6))
plt.plot(data, linewidth=0.2)
pdf.savefig(f, bbox_inches="tight")

f  = plt.figure(3, figsize=(12, 6))
working_data, measures = hp.process(data, 100, windowsize=1, freq_method='welch', calc_freq=True)
p  = hp.plotter(working_data, measures, show=False)
pdf.savefig(f, bbox_inches="tight")

f  = plt.figure(4, figsize=(12, 6))
data = data[int(len(data)/3):]
working_data, measures = hp.process(data, 100, windowsize=1, freq_method='welch', calc_freq=True)
p  = hp.plotter(working_data, measures, show=False)
pdf.savefig(f, bbox_inches="tight")

f  = plt.figure(5, figsize=(12, 6))
data = data[int(len(data)/3):]
working_data, measures = hp.process(data, 100, windowsize=1, freq_method='welch', calc_freq=True)
p  = hp.plotter(working_data, measures, show=False)
pdf.savefig(f, bbox_inches="tight")

f  = plt.figure(6, figsize=(12, 6))
data = data[int(len(data)/3):]
working_data, measures = hp.process(data, 100, windowsize=1, freq_method='welch', calc_freq=True)
p  = hp.plotter(working_data, measures, show=False)
[... 3323 lines cut ...]
105.91117683751277
106.22724168389608
106.39193974019352
105.81177425947324
105.65885920693759
105.78477295233336
105.59338642990082
105.5367464716324
105.4792799167018
105.54162111115
105.35223589399624
105.45637547380562
105.45589396600363
105.22920962443548
105.05620426862303
105.13673948551269
105.172991914582
105.03344133909106
105.04418975218837
104.8999760376271
105.00111380840158
104.9003176893209
104.76588640572191
104.43106774131103
1

paulvangentcom, 2019-01-28:
Can  you send the RAW.txt file to info@paulvangent.com? I cannot open it from here.

Cheers,
Paul

paulvangentcom, 2019-02-05:
Yes I see what's happening. It tries to minimize certain parameters globally. Usually a process that gives good results by default, except when a large enough portion of the signal is noisy like here. I will flag this since the current method needs to be refined and I need to allot some time for this soon.

For now a solution is to use the built-in peak enhancer `enhance_peaks` and `butter_lowpass_filter` filtering methods:

```python
import heartpy as hp

data = hp.get_data('RAW.txt')
filtered = hp.butter_lowpass_filter(data, 5, 100.0, 3) #third order butterworth with cutoff @ 5Hz
filtered_and_scaled = hp.scale_data(filtered)

wd, measures = hp.process(filtered_and_scaled, 100.0, reject_segmentwise=True, calc_freq=True)

hp.plotter(wd, measures)
```

Yielding:
[image]

As mentioned it is a suboptimal solution that throws away a portion of the signal. I will close this issue and make it a point on the to-do to refine the methods.

closed 2019-02-06 by paulvangentcom