paulvangentcom/heartrate_analysis_python #9: Integrate PPG methods with NeuroKit

paulvangentcom, 2019-03-07:
Hi Renatosc,

I'd heard of it but have not used it so far. The repo looks interesting indeed! Right now I'm working on adding an ECG-specific peak finder to HeartPy based on an often used template matching method.

I'll have a look later today at NeuroKit. The integration between our two packages should be relatively simple. Might indeed be nice to use this to add PPG capability to NeuroKit :)

Cheers,
Paul

_Originally posted by @renatosc in https://github.com/paulvangentcom/heartrate_analysis_python/issues/7#issuecomment-470398578_

renatosc, 2019-03-07:
Interesting you to mention about working on ECG because that is exactly what I was looking for now (I just finishing putting the PPG continuous code to work). Since you mention it in your tutorials, I got the impression that you already had ECG also implemented.

I have done some contributions to NeuroKit, I would gladly help in the PPG integration.

paulvangentcom, 2019-03-07:
Yes there current peak detection iteration works with ECG as well, even though it was originally intended for PPG waveforms it's a general method that translates quite well.

However there will always be cases where it fails, which is why I want to get an alternate (backup) method in case the variable thresholding fails. The same is true vice versa: a template matching method can strengthen cases where detection on PPG fails as well.

Keep an eye on the repo. If things go well today I'll finish it over my lunch break.

paulvangentcom, 2019-04-24:
closed due to a lack of time for now.

closed 2019-04-24 by paulvangentcom