paulvangentcom/heartrate_analysis_python #17: ValueError: min() arg is an empty sequence

vishakha18tiwari, 2019-03-31:
Hey. I am running this code but it is throwing error.
import pandas as pd
import heartpy as hp
df = pd.read_csv('E:/Autistic_data/drive03'+'.csv')
working_data, measures = hp.process(df.iloc[:,5], 100, calc_freq=True)

ValueError Traceback (most recent call last)
in 
1 df = pd.read_csv('E:/Autistic_data/drive4'+'.csv')
----> 2 working_data, measures = hp.process(df.iloc[:,5], 100, calc_freq=True)

~\AppData\Roaming\Python\Python37\site-packages\heartpy\heartpy.py in process(hrdata, sample_rate, windowsize, report_time, calc_freq, freq_method, interp_clipping, clipping_scale, interp_threshold, hampel_correct, bpmmin, bpmmax, reject_segmentwise, measures, working_data)
647 working_data['hr'] = hrdata
648 rol_mean = rolmean(hrdata, windowsize, sample_rate)
--> 649 working_data = fit_peaks(hrdata, rol_mean, sample_rate, working_data=working_data)
650 working_data = calc_rr(working_data['peaklist'], sample_rate, working_data=working_data)
651 working_data = check_peaks(working_data['RR_list'], working_data['peaklist'], working_data['ybeat'],

~\AppData\Roaming\Python\Python37\site-packages\heartpy\heartpy.py in fit_peaks(hrdata, rol_mean, sample_rate, bpmmin, bpmmax, working_data)
377
378
--> 379 working_data['best'] = min(valid_ma, key=lambda t: t[0])[1]
380 working_data = detect_peaks(hrdata, rol_mean, min(valid_ma, key=lambda t: t[0])[1],
381 sample_rate, update_dict=True, working_data=working_data)

ValueError: min() arg is an empty sequence

Below is the link for dataset I have used.

https://drive.google.com/open?id=1e08C3kyeq0kkuFdhQ9J7a0vUsZRGY0D8

paulvangentcom, 2019-04-01:
Thanks for sharing the data. I've requested access to the data through my gmail (palkab29@gmail.com) account, as right now I cannot access it.

- Paul

paulvangentcom, 2019-04-02:
The analysis runs fine on my end.

You need to make sure you set the right sample rate when calling `process()`. Right now you're passing the default 100.0. The file you provided seems to be 31 Hz. Also scale the data prior to the analysis, right now it's in mV.

Consider an example:

```python
import heartpy as hp
import pandas as pd

df = pd.read_csv('drive03.txt')
data = hp.scale_data(df['ECG'].values) #we need to scale the data as its ECG

timer = df['Elapsedtime'].values * 1000 #get the timer values and convert to ms as its in seconds
fs = hp.get_samplerate_mstimer(timer) #compute sample rate

wd, m = hp.process(data[0:2000], sample_rate = fs) #run analysis on first 2000 points
hp.plotter(wd, m) #visualise
```

[image]

closed 2019-04-02 by paulvangentcom

vishakha18tiwari, 2019-04-02:
Hey. The code has successfully executed for all the datasets which were showing the above error previously. But the datasets which were showing output successfully previously are showing the same error now. Example of this dataset is

https://drive.google.com/file/d/1_Z4KJxPoruijFqQ8EboFfKCRU26xDWWR/view?usp=sharing

paulvangentcom, 2019-04-02:
Be sure to always visualise the data you're attempting to analyse. The data file you provided contains a different ECG waveform (likely from a different electrode) than the other one provided:

[image]

If this is the case ensure you flip the signal first. There's a method in the package `raw_to_ecg()` to do this:

```python
data = hp.scale_data(df['ECG'].values) #get the signal into an array and scale it
data = hp.raw_to_ecg(data) #flip the signal

#again determine sample rate
timer = df['Elapsed time'].values * 1000
fs = hp.get_samplerate_mstimer(timer) #apparently in this file it's 15Hz.

wd, m = hp.process(data[0:2000], fs) #again run analysis on first segment 

hp.plotter(wd, m) #visualise
```

gives:

[image]

paulvangentcom, 2019-04-02:
Make sure you visualise the results and inspect them as well, don't ever trust any package blindly!

-Paul

vishakha18tiwari, 2019-04-06:
Hey. Paul. Thanks for the help. Everything is working now. I would like to ask is process method can be used for PPG data directly as i am doing project using pulse sensor  ?