paulvangentcom/heartrate_analysis_python #19: Accessing IBI intervals

FrancoisCosta, 2019-04-05:
Hi Paul!
Great work with this toolbox :) I've been using it with my Pulse Sensor and seem to be getting some very good results on the large part.
However, I have had a couple of problems while attempting to collect some results:

1 - List of IBIs from detected peaks
I have run the peak detection functions 'hp.enhance_peaks' and 'hp.process' functions on my data. I would now like to access the list of IBI values calculated from all the consecutively accepted peaks (provided no peaks were rejected in between the accepted peaks). However when accessing measures['RR_list_cor'] and measures['RR_list'] all the values produced seem to be rounded to the nearest 20mS (values such as 860, 880, 900 etc..). Would it be possible to the nearest mS instead? 
[image]

2- Varying signal amplitude
My sensor signal amplitude varies quite a lot over time. To correct this, I used the enhanced_peaks functions which improved the results. However, I am still having some issues with incorrectly rejected peaks and undetected peaks. Images included below. Are there any other functions I could be using to rectify this? (I have included a screenshot of both a well-performing sample and one where the problems occur below)
[image]
[image]

Thanks for the help! I look forward to using this toolbox and hopefully contributing in the future :)

Francois

paulvangentcom, 2019-04-07:
Hi Francois,

Thanks for the kind words. Can you share the data behind the two plots you place with me? I am working on an extension to the `enhance_peaks()` function that will normalise locally rather than globally. This would give me a chance to test it.

paulvangentcom, 2019-04-07:
Additionally, take a look at the heartrate arduino repo here, especially the AVR simple logger with scaling: https://github.com/paulvangentcom/heartrate_analysis_Arduino/tree/master/implementations/AVR/SimpleLogger_Scaling_AVR

Docs here: https://embedded-heart-rate-analysis-toolkit.readthedocs.io/en/latest/implementations.html#simple-logger

This is an implementation for a PPG sensor like the Pulse Sensor you use, plus Arduino. It auto-scales the data as it is recording. This might help you achieve what you need without using the `enhance_peaks()` in heartpy.

paulvangentcom, 2019-04-07:
Regarding the RR-intervals. I assume your sample rate is 50Hz. The module finds the nearest peak. I'm working on a solver to interpolate each peak and find the maximum that way, which could be more accurate.

For now I recommend upping your sample rate to 1000Hz for 1ms R-R accuracy. The linked Arduino sketch can reach those sampling speeds easily.

-Paul

FrancoisCosta, 2019-04-07:
Hi Paul, thanks for the reply!

I've attached the csv file with the Pulse Sensor Data I've been using. It was sampled at 50Hz over a period of approximately 15mins. 
Let me know if you would like longer datasets, I should have several hours in the next couple of weeks (I aim to collect data to analyse sleep). 

[PulseSensorData_vanGent.txt](https://github.com/paulvangentcom/heartrate_analysis_python/files/3051438/PulseSensorData_vanGent.txt)

The first graph is believe represents the [:15000] samples whereas the second graph represents [30000:]

Regarding the RR intervals - ah I see! I'll give that Arduino sketch a go tomorrow morning and let you know how it goes (I don't have access to the sensor today unfortunately). 

Please let me know if/when the local enhance_peaks() function and the IBI interpolated solver get implemented :) would be of great help for my project.

paulvangentcom, 2019-04-08:
I'll push the peak position interpolation step up my list then and hope to get to it this week. I'll keep you updated.

-Paul

FrancoisCosta, 2019-04-08:
Perfect! Thanks :)

FrancoisCosta, 2019-04-17:
Hi Paul, any update on this?

Thanks,
FB

paulvangentcom, 2019-04-18:
Hi Francois,

Not yet I'm afraid. I need to find the time to fit heartpy work in between my phd work so it doesn't always work out...

I hope this weekend I can have a look.

-Paul

paulvangentcom, 2019-04-19:
update, I've finished the interpolation method, which upsamples and re-estimates peak position. On the first peak in your signal it looks like this:

[image]

Or, for the first 10 seconds of your signal you can go from 20ms accuracy (50Hz):
`array([ 280., 1160., 2080., 2980., 3860., 4740., 5620., 6480., 7300., 8120., 8920., 9700.])`

to peaks estimated to 1ms accuracy (1000Hz)
`array([ 276., 1166., 2079., 2984., 3865., 4737., 5622., 6477., 7295., 8111., 8914., 9692.])`
(or of course higher if you want, but I don't see why)

You can specify the precision you want and the method does the rest. I need to test the implementation and update the docs. I'll probably push the update to github today or tomorrow. Just wanted to keep you informed.

FrancoisCosta, 2019-04-19:
Excellent work! Thanks Paul! 
I look forward to giving it a go :)
Do you foresee any negative impact on the accuracy of the peak location when upsampling and re-estimating?

paulvangentcom, 2019-04-23:
Please pull the latest version, I've implemented the above described interpolation to the `process()` and `process_segmentwise()` functions. Usage is simple:

```python
import heartpy as hp

data = hp.get_data('data.csv')

#the 'high_precision' flag sets the mode as active
#the 'high_precision_fs' variable sets the target sample rate to upsample to
#default is 1000.0 Hz, meaning 1-ms peak accuracy
wd, m = hp.process(data, sample_rate = 100.0, high_precision = True, high_precision_fs = 1000.0)

print(wd['RR_list'])
```

Gives

`[1019.  982.  968.  998. 1051. 1092.  982.  909.  892.  952. 1085. 1157. 1130. 1023. 1041. 1067. 1046.  944.  963. 1035. 1094. 1012.  980.]`

Good luck!

- Paul

closed 2019-04-23 by paulvangentcom