paulvangentcom/heartrate_analysis_python #22: Breathing rate return "nanHz"

kmf909, 2019-04-11:
hi Paul
i try to estimate respiration rate using data from link below. but it return nanHz breathing rate. any suggestion to overcome this.

http://peterhcharlton.github.io/RRest/synthetic_dataset.html

---------------CODE start--------------
import heartpy as hp
data = hp.get_data('rrest-syn001_data.csv', delim=',', column_name='ppg')
fs = 500
working_data, measures = hp.process(data, fs, report_time=True)
print('breathing rate is:%sHz' %measures['breathingrate'])

paulvangentcom, 2019-04-11:
Hi kmf909!

The currently implemented breathing rate extraction is relatively simple (an extension is planned, though as of now BR extraction is not a main feature of HeartPy). As of now it looks at frequency modulation caused by breathing, as this is what we mainly encounter in our PPG datasets collected in the field. As such, only the 'FM' signals from the synthetic dataset are usable. 

As an example, I'll take the last file in the data set your linked ('rrest-syn192'), which is noted to have FM and:
```
Sampling frequency: 500 Hz
Respiratory modulation type: FM
Simulated respiratory rate: 20 breaths/min
Simulated heart rate: 200 beats/min
```

Consider this standard code blob:

```python
import heartpy as hp

data = hp.get_data('rrest-syn192_data.csv')
ppg = data[:,0] #first column is ppg signal

#Analyse it. Note I pass the bpmmax argument, as the default range is 40 < BPM < 180
#The package will raise a BadSignalWarning otherwise, assuming something is wrong with the signal.
wd, m = hp.process(ppg, 500.0, bpmmax=240)

m['breathingrate']
```

This gives ```0.33809845808055317```(Hz), which translates to one breath cycle per 3 seconds, or ~20 breaths a minute as specified in the file.

And the first half minute of data visualised:
[image]

-Paul

paulvangentcom, 2019-04-11:
As a side-note, the actual _reason_ you see NaN in breathingrate in the other segments is because the RR-intervals have zero variance, so the rest of the computation fails as well.

closed 2019-04-11 by paulvangentcom

kmf909, 2019-04-11:
Hi Paul. Thank you for the prompt reply. crystal clear explanation.

paulvangentcom, 2019-04-11:
Happy to help!