paulvangentcom/heartrate_analysis_python #24: Plot visualisation when using hp.process_segmentwise

FrancoisCosta, 2019-04-21:
Hi Paul,

When using the hp.process_segmentwise() function I have not been able to figure out how to plot separate graphs for each segment (other than separating the data beforehand).
Is there a way to do this?

Cheers!

paulvangentcom, 2019-04-23:
How would you expect this functionality to behave? 

I like the idea but I'm having some trouble seeing exactly what a good implementation would be. For example. if there are 100 segments, you wouldn't want 100 plots puked across the screen.

Let me know if you have ideas.

- Paul

FrancoisCosta, 2019-04-24:
Hi Paul,

Good point! Perhaps the graphs could be saved to a new folder within the directory rather than being displayed? With each graph being named using something like: 

`for i in range(len(working_data_segment['segment_indices'])):`
`plt.savefig('folder_name/graph_title_(%d).png' % i, bbox_inches='tight')`

A useful feature would be to be able to input a range of segments that you would like to be saved (and potentially the step) 

`def save_segment_graphs (..., start=20, end= 40, step=2)`

Would give the graphs for segments (20, 22, 24 ... 38, 40)

Let me know what you think :)

Francois

paulvangentcom, 2019-04-24:
Alright that is a good idea for the functionality. Let me see what I can implement

paulvangentcom, 2019-04-24:
Please pull latest version and see the docs for [full documentation here.](https://python-heart-rate-analysis-toolkit.readthedocs.io/en/latest/quickstart.html#plotting-results-of-segmentwise-analysis)

Example with data-file from repo:

```python
import heartpy as hp

data = hp.get_data('data3.csv', column_name='hr')
data = hp.filtersignal(data, cutoff=[0.5, 3.5], sample_rate=100.0, order=3, filtertype='bandpass')
data = hp.scale_data(data)

#process it
wd, m = hp.process_segmentwise(data, 100.0, segment_width=60, segment_overlap=0.5, high_precision=True)

#use new function
hp.segment_plotter(wd, m, path='resulting_plots')
```

paulvangentcom, 2019-04-24:
Above code results in this by the way:

[image]

closed 2019-04-24 by paulvangentcom