paulvangentcom/heartrate_analysis_python #25: process_segment_wise not working since update

FrancoisCosta, 2019-05-01:
Hi Paul,

I seem to be having some serious problems with the process_segment_wise since the latest update...

**dataset**:
[01-05-19_00-25-52.txt](https://github.com/paulvangentcom/heartrate_analysis_python/files/3134482/01-05-19_00-25-52.txt)

I have a dataset of which I'm using the first 500000 readings. My sampling speed is 1000Hz so that should correspond to the following segments when running:

`working_data_segment, measures_segment = hp.process_segmentwise(data, sample_rate=1000.0, segment_width = 300, segment_overlap = 0.9)` 

**Expected results:**
[(0, 300000),(30000, 330000),(60000, 360000),(90000, 390000),(12000, 420000),(150000, 450000),(180000, 480000),(210000, 500000)

**However - it produces the following segments:**

[(0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (0, 300000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (29999, 330000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (59999, 360000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (89999, 390000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999, 420000), (119999,420000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (149999, 450000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (179999, 480000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000), (209999, 500000)]

It worked well before the latest update, not sure what's going wrong now. I tested it on another dataset and got similar results

paulvangentcom, 2019-05-06:
hm that shouldn't happen indeed. Let me have a look

paulvangentcom, 2019-05-06:
Seems I was a bit sloppy with an indentation, thanks for catching the error. It works as expected now.

What happened is that for each dictionary key an additional index range was appended, rather than just for each segment.

closed 2019-05-06 by paulvangentcom

FrancoisCosta, 2019-05-21:
Hi Paul,

This seems to be a problem again for me unfortunately.

paulvangentcom, 2019-05-23:
Make sure you get the latest version from the github, uninstall the old one first just to be sure:

```pip uninstall heartpy```

then 

```python setup.py install``` from the cloned folder from git.

Using what's on github + your file above I cannot reproduce the error:

[image]

the behaviour seems as intended.