paulvangentcom/heartrate_analysis_python #35: attribute missing

abdulkhaliq7, 2019-10-03:
hi, I have downloaded in a any way possible and still I' getting error:

 "filtered = hp.enhance_ecg_peaks(hp.scale_data(ecg), sample_rate, iterations=4, aggregation='median', notch_filter=True)

AttributeError: module 'HeartPy' has no attribute 'enhance_ecg_peaks' ''

I will appreciate your help!!!!

paulvangentcom, 2019-10-08:
Make sure you uninstall the older version first prior to installing the new one, it seems you're still not on the newest version..

```pip uninstall -y heartpy```

abdulkhaliq7, 2019-10-08:
thank u v. much it is working.

I'm doing heart disease detection using deep-learning,,,how do I download MIT-BIH arrhythmia database and loaded with the wfdb package......and is the heartpy support heartbeat segmentation and feature extraction. 

thanks in advance.

closed 2019-10-08 by abdulkhaliq7