paulvangentcom/heartrate_analysis_python #36: Multiple channels?

sknepal, 2019-11-01:
Hi,

Thank you for this awesome library.

I was wondering if there's a way to analyze multiple channels. I've seen some papers where linear combinations, averages of different channels are taken to reduce noises or motion artifacts. For instance when recording raw ppg from smartwatches, or arm bands such as Polar OH, we get values from 3 channels. Is there anything in the library that handles such multi-channel inputs?

I'd appreciate if you could please suggest what's currently the go-to approach for handling raw ppg signals from different channels? Visualizing the signals and only using the less noisy one or averaging all or something else altogether. Any insight into this would really be helpful.

Thank you!

paulvangentcom, 2019-11-05:
Hi Sknepal,

Can you share a few of the papers? I have some ideas on how to approach this but it may be good to look at the literature first.

paulvangentcom, 2019-11-08:
Also, is it possible to share a data file with the three-channel recording you mention? I'd like to try a few things.

Paul

signekb, 2019-11-08:
Hi Paul, 
We are working on some ECG data with three channels and were wondering the same as sknepal. Is there a way to analyse multiple channels using your toolkit? 

Thank you for in advance! and for a great package :-) 
/SigneKB

signekb, 2019-11-08:
We cannot share the data, unfortunately, since we are using data from the MAHNOB HCI tagging database (https://mahnob-db.eu/hci-tagging/) which you'll need to sign an EULA to get access to. However, if you sign the EULA and send it to them, you can get access. 
/SigneKB

paulvangentcom, 2019-11-08:
Is the data tagged with which channel corresponds to which electrode? Usually V4, V5 or V6 works well in analysis.

I'll have a look over the weekend for multi-channel integration and whether it makes sense to put in

signekb, 2019-11-08:
The channels are called: 
EXG1 - position: ECG1 (upper right corner of chest, under clavicle bone) 
EXG2 - position: ECG2 (upper left corner of chest, under clavicle bone)
EXG3 - position: ECG3 (left side of abdomen) 

Source is their data manual:
https://mahnob-db.eu/hci-tagging/media/uploads/manual.pdf

Thanks!

paulvangentcom, 2019-11-08:
Ok, that is the standard three-lead setup. Plot a short 10 sec segment for each channel. You'll likely see something like: [(source)](https://www.biopac.com/application-note/ecg-ekg-electrocardiography-12-6-3-lead/)

[image]

For now I suggest you take the channel displaying the clearest QRS complex (top row in example image) and use that for the analysis.

I will update you if I manage to implement mutli-channel capability.

closed 2020-10-22 by paulvangentcom