paulvangentcom/heartrate_analysis_python #40: not able to use Notch filter

kmf909, 2019-12-17:
hi Paul

im using heartpy v1.2.6. i try to apply notch filter using signal_filter and remove_baseline_wander. the code as below. i attached sample data (1.txt). when i apply lowpass filter the code run without issue but when i try to apply notch filter using remove_baseline_wander function it pop up error as below

[image]

when i try to apply notch filter using signal_filter function the error as below

[image]

`import numpy as np
import heartpy as hp
import matplotlib.pyplot as plt

input = '1.txt'
dataPPG = numpy.loadtxt(input, delimiter=",", usecols=0)
plt.plot(dataPPG, label='Raw_data')
plt.legend()
plt.show()

########################baseline wander###########################
bw = hp.remove_baseline_wander(dataPPG, sample_rate= 100, cutoff=60.0)
plt.plot(bw, label='Raw_data')
plt.show()`

########################## signal_filter##########################
nf = hp.filter_signal(dataPPG, cutoff=60,sample_rate=100,filtertype='notch')
plt.plot(nf, label='notch filter')
plt.show()

#########################signal_filter_lowpass####################

lp = hp.filter_signal(dataPPG, cutoff=0.05,sample_rate=100,filtertype='lowpass')
plt.plot(nf, label='Lowpass filter')
plt.show()

Thanks in advance.

[1.txt](https://github.com/paulvangentcom/heartrate_analysis_python/files/3972619/1.txt)

paulvangentcom, 2019-12-17:
What version of SciPy are you using? You need at least SciPy 1.2.0. 

Upgrade SciPy to the latest and the problem should go away.

paulvangentcom, 2019-12-17:
p.s. as of now this is an incompatibility between Python 2.7 and HeartPy that I won't address, as official 2.7 support will be dropped anyway in January 2020.

closed 2019-12-23 by paulvangentcom