paulvangentcom/heartrate_analysis_python #41: scale_sections function bug

Lauren-ED209, 2020-02-10:
Dear HeartPy team,

I am a maker. I am designing a PPG wearable device for real-time HRV analysis, data collecting and visualization. 

When developing the PC application, I integrated your library. When working on preprocessing for rolling data, I find that "scale_sections" function issue may analysis failed randomly. 

**Let me share what I've found:**

`output = np.empty(len(data))`
// length of output array is the same as length of data array. 

so if data_end is not equal to length of data. Then some data at the end of output numpy array will not be assigned. They will be random value. If one of them is nan, then the process fuction will crash. 

**To fix this:**
just change return np.array(output) -> return np.array(output[0:data_start])

Let me know if I described the issue clearly. 
Feel free to let me know your thoughts. 

Best,
Lauren

Lauren-ED209, 2020-02-10:
BTW, I've been thinking about opensource the PPG wearable device to the market. Let me know if it's valuable to have a opensource hardware completely compatible with heartpy and supported by heartpy. People can make and customize the device according to their needs based on our design. 

I can share more information with u including photos and design files for our deeper discussion. 

Best,
Lauren

adamhub, 2020-03-14:
Hey @Lauren-ED209 
I am making a wearable PPG device too and came across this issue. I fixed it by fixing my import to avoid float rounding issues by just using integers when making my CSV via numpy.
Would love to hear more about your project. I ended up with the MAX30102 sensor and am about to try it with the Nano RF board from Emakefun. It contains a NRF24L01 radio rather than wifi, which suites my purposes better. 

@paulvangentcom KILLER work. Thank you so much!

paulvangentcom, 2020-03-15:
@Lauren-ED209 Yes np.empty just assigns memory space and doesn't fill it, so if the array isn't filled, whatever value was at that specific byte in memory will be in the array. 

I did not realize there were cases where your described issue popped up. I'll update the function. Was thinking about removing scale_sections altogether as in my experience it doesn't improve quality of analysis anyway.

An open source product compatible with HeartPy is of course always welcome, it's one of the reasones I open sourced the work in the first place : )

Cheers

paulvangentcom, 2020-03-15:
@Lauren-ED209 @adamhub 

If either of you needs help with pcb design or embedded C for the arduino projects I can lend a hand as well.

closed 2020-03-15 by paulvangentcom

Lauren-ED209, 2020-04-23:
@paulvangentcom, would you mind send me an email to me? I've finished a product design based on this open source project! 
Would like share more detail via email to u! 

@adamhub, Cool! Would like to exchange some ideas and product details via email! I am using zigbee cc2530, which is capable of extending distance between receiver and transimitter side without limit. And I've finished an industrial design on it.

Email:  ed209.lauren@gmail.com

Check out product images here: 
https://imgur.com/h8lx1pQ
https://imgur.com/azbjOAb
https://imgur.com/q6SxSU1
https://imgur.com/VeQvF8t

commit b9bb5c27 (2020-03-15): close #41

Add suggested change by @Lauren-ED209.

Mark scale_sections as deprecated as it will be removed/replaced in a future version.