paulvangentcom/heartrate_analysis_python #42: Sample Rate for Frequency Domain Calculation

lzwang26, 2020-04-06:
Dear Paul:

I'm not sure whether this is an issue or not, but I found that in the function calc_fd_measures(), there no sample rate as a input variable. And for all three methods (fft, periodogram and welch), the sample rate is fixed as 1000. Could you please tell me the reason for this?

Thanks so much for your help!

paulvangentcom, 2020-04-07:
calc_fd_measures() and calc_ts_measures() both take already detected peak-peak intervals as input values to compute their statistics from, so no sample rate is required anymore for the time-series measures, as they describe the variation between peaks.

For the frequency domain measures, which describe the peak-peak variation over time in the frequency domain, intervals are placed on a ms-based timeline and interpolated. Since the base unit of the intervals is ms, doing so results in a timeline with 1-ms increments: 1000Hz.

Hope this answers your question

Cheers,
Paul

Cheers
Paul

closed 2020-04-07 by paulvangentcom

lzwang26, 2020-04-09:
Dear Paul:

Thanks so much for your explanation! I've noticed that you use the rr_list to be the pointer and then the sample rate should be 1000. So doesn't it mean the sample rate of the PPG signal will not influence the frequency domain result?

Thanks!

lzwang26, 2020-04-09:
Dear Paul:

Sorry for asking again. I see there's an input called square_spectrum in the function calc_fd_measures. The default is True, but this input is never used in the function. Could you please tell me what this for? And what this will influence the result?

Thanks so much!

paulvangentcom, 2020-04-09:
> Dear Paul:
> 
> Thanks so much for your explanation! I've noticed that you use the rr_list to be the pointer and then the sample rate should be 1000. So doesn't it mean the sample rate of the PPG signal will not influence the frequency domain result?
> 
> Thanks!

The frequency measures are computed on peak-peak intervals, which have nothing to do with the original PPG signal sample rate.

paulvangentcom, 2020-04-09:
> Dear Paul:
> 
> Sorry for asking again. I see there's an input called square_spectrum in the function calc_fd_measures. The default is True, but this input is never used in the function. Could you please tell me what this for? And what this will influence the result?
> 
> Thanks so much!

Indeed, it seems that did not move correctly to the new refactor. It is because in some cases you'd want to square the frequency spectrum. I'll put that in an issue and get on it, thanks.