paulvangentcom/heartrate_analysis_python #49: Frequency Analysis Result

iPsych, 2020-09-07:
First, I sincerely appreciate for the properly working BVP/ECG analysis codes you shared.
I have a question about the frequency domain result.
When I try to analyze BVP data with very long length (>3hrs),
The result shows EXTREAMLY large value for hf and lf power (line 44).
But the lf/hf makes sense, and the value is reasonable if I divide it by 1000^2 (maybe for ms^2).
Would you let me know what is the unit of raw HF, LF and is it correct to divide it by 1000^2 for get the trivial ms^2 result?
[image]

paulvangentcom, 2020-09-24:
The result is raw area-under-curve for the frequency range, so you need to transform it into your desired metric such as ms^2

Hope this helps

Cheers
Paul

closed 2020-09-24 by paulvangentcom

iPsych, 2020-09-26:
@paulvangentcom 
Thanks. I am little confused what 'raw area-under-curve for the frequency range' means.
So, to convert ms^2 metric, should I divide it with (1000*1000)? Or the equation should be changed based on my sampling rate?