paulvangentcom/heartrate_analysis_python #57: plot_poincare() method error

Franziska-L, 2021-02-24:
I'm getting an error when I try to call the method plot_poincare(): "alpha must be a float or None".
This error also occurs in the examples like 2_regular_ECG and 5_noisy_ECG of the Github page.

Issue #44 seems to have fixed this but I still get the error.
My HeartPy Version is 1.2.6

paulvangentcom, 2021-02-26:
Try uninstalling first

```pip uninstall heartpy```, then cloning the repo and running ```python setup.py install```

The pip release is a bit behind, it's on my radar to fix. Of course feel free to let me know if the problem persists. 

Cheers

Franziska-L, 2021-03-02:
Thanks. Poincare_plot() is working now but I get another error: `UserWarning: Matplotlib is currently using module://ipykernel.pylab.backend_inline, which is a non-GUI backend, so cannot show the figure.`

I'm using Python 3.7 since 2.7 is deprecated.

paulvangentcom, 2021-03-02:
This error has to do with Jupyter Notebook and not HeartPy.

Try adding ```%matplotlib inline``` to the notebook cell where you do the imports.

If that doesn't work, have a look at the suggestions here https://stackoverflow.com/questions/37365357/when-i-use-matplotlib-in-jupyter-notebook-it-always-raise-matplotlib-is-curren

Cheers

paulvangentcom, 2021-03-14:
I'm closing this for now. Feel free to reopen or open another issue if you encounter further problems!

Cheers
Paul

closed 2021-03-14 by paulvangentcom