paulvangentcom/heartrate_analysis_python #58: BPM of every R peaks and time of this bpm?

Mark462246, 2021-02-27:
Hi!
I'm a university student. In my dissertation, an ECG displaying and processing program must be created. Something similar as in this video:
https://www.youtube.com/watch?v=9jKccn9JaGc

I found the HeartPy yesterday and thought the HeartPy could help with my work.
There was already a similar question: https://github.com/paulvangentcom/heartrate_analysis_python/issues/7
But I’m not sure the solution there is right for my problem.  The solution to that problem has been the proccess_segmentwise method,  which is known to split the ecg into segments and the width of the segment is given in seconds.

The BPM value is calculated from two RR peaks.  So it's okay to split it into segments based on seconds using the process_segmentwise () method, but this only gives the bpm of the segments. I want to get a bpm value after each RR peak. As in this picture:
[image]

After all, I want to split it segments by RR peaks and get the bpm values of these RR segments and at what point in time I get these bpm values, which in this case is equal to at what point in time a peak R occurs.

I hope understable what I would like to do. I don't know if this is feasible.

Another problem is that the segment_plotter () method doesn't work for me, but I'm sure I'm setting something wrong.
I tried with this example: 
```
import heartpy as hp

data, timer = hp.load_exampledata(2)
sample_rate = hp.get_samplerate_datetime(timer, timeformat='%Y-%m-%d %H:%M:%S.%f')
working_data, measures = hp.process_segmentwise(data, sample_rate, segment_width = 40, segment_overlap = 0.25)

hp.segment_plotter(working_data,measures)
```

And I get this error:
```
Traceback (most recent call last):
  File "F:/.../.../.../01.py", line 7, in <module>
    hp.segment_plotter(working_data,measures)
  File "F:\...\...\...\heartpy\visualizeutils.py", line 198, in segment_plotter
    p = plotter(wd_segment, m_segment, show=False)
  File "F:\...\...\...\heartpy\visualizeutils.py", line 80, in plotter
    fs = working_data['sample_rate']
KeyError: 'sample_rate'
```

paulvangentcom, 2021-03-02:
Hi Mark,

Thanks for reaching out. My reply will be brief as I broke my hand. Not much typing now.

Usually in these devices you get the BPM for the last period (say: last 10 peaks or last 30 seconds). Heart rate speeds up and slows down considerably with each beat. This is the main reason you don't want to compute BPM based on 1 peak-peak interval. 

The best way to go here is to implement a circular buffer, and every time interval (say 1 sec) you run HeartPy over the buffer. This will give you the BPM of the time interval equal to the buffer size. This way you can update your parameters as frequently as you like.

In the movie you linked they seem to do a similar thing, BPM updates every few seconds.

If you need pointers I'd be happy to whip up some example code. Let me know.

Cheers,
Paul

paulvangentcom, 2021-03-02:
I'll look into the segment_plotter. Thanks for raising that.

paulvangentcom, 2021-03-20:
the segment plotter issue has been fixed in 9adb10d, be sure to uninstall heartpy then update.

paulvangentcom, 2021-03-20:
Closing this due to inactivity. Feel free to re-open if you need further assistance.

Cheers,
Paul

closed 2021-03-20 by paulvangentcom