paulvangentcom/heartrate_analysis_python #59: some output measures are different in v1.2.6

handlercc, 2021-02-28:
Hi! I was running through the smart watch data and following your Analysing Smartwatch Data notebook but some of the output values differ a lot such as the sdsd and the rmssd, hoping there could be some analysis as to why or what these numbers would mean. 

From the notebook:
bpm: 78.981709
ibi: 759.669563
sdnn: 79.989043
sdsd: 99.586554
rmssd: 99.586688
pnn20: 0.385797
pnn50: 0.205374
hr_mad: 41.049817
sd1: 36.948123
sd2: 73.723698
s: 8557.548377
sd1/sd2: 0.501170
breathingrate: 0.239659

My output:
bpm: 75.543450
ibi: 794.244901
sdnn: 92.544456
sdsd: 33.387928
rmssd: 49.947643
pnn20: 0.644444
pnn50: 0.244444
hr_mad: 35.056933
sd1: 34.389733
sd2: 85.202755
s: 9205.180571
sd1/sd2: 0.403622
breathingrate: 0.200000

paulvangentcom, 2021-03-02:
Thanks for letting me know, strange indeed I see the same with a newer version of HeartPy.

Something to dig into, will keep you updated.

-Paul

paulvangentcom, 2021-03-20:
Turns out some improvements in the rejection of peaks, as well as some changes to how plotting handles figures led to strange behaviour. I've updated the notebook to reflect these changes now. 

In some areas HeartPy has become stricter. It probably detected quite some false positives in the earlier version, leading to an inflation of the HRV metrics.

- Paul

closed 2021-03-20 by paulvangentcom