paulvangentcom/heartrate_analysis_python #62: Unique data

mshafieluru, 2021-03-19:
Hi Paul,

Nice articles and source code. Is the heart data unique for each user?

Please explain which data points are unique for each user.

Regards
Shafi

paulvangentcom, 2021-03-20:
I'm not sure what you mean. There's no duplicate data all data was recorded on participants (e.g. no synthetic data), if that's what you're looking for.

- Paul

mshafieluru, 2021-03-20:
Hi 

That was not my intention. I fully agree that it's authentic data. 
My question is what attributes in the collected dataset are unique for each user ? or a custom logic is developed to determine/identify a user.

Can this data be used for the biometric identification of a user?

paulvangentcom, 2021-03-20:
Ah like that. In the provided exampledata examples 0 and 1 are the same person, example 2 is a different person.

The data used in the papers for validation is partially from one of our experiments, and from an online dataset that is linked in the paper. These have unique data per participant.

Cheers,
Paul

mshafieluru, 2021-03-20:
Thanks. This helps

closed 2021-03-20 by mshafieluru