paulvangentcom/heartrate_analysis_python #63: Only working fine on the 14500:20500,

Jaswanth-Devarinti, 2021-03-20:
It's only working on the range given, for other ranges, it's not working.

[image]

paulvangentcom, 2021-03-20:
Please provide the full code and data set you're using. There's little I can do with just this info

-Paul

Jaswanth-Devarinti, 2021-03-20:
Same dataset and same code, I will share the code 5 min

Jaswanth-Devarinti, 2021-03-20:
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

These are the snippets in the code.

paulvangentcom, 2021-03-20:
Can you share the .py file? I cannot see the images. Don't know what set the 'same dataset' is

Jaswanth-Devarinti, 2021-03-20:
Sure, here it is.

https://colab.research.google.com/drive/15A_98W1lsyXFiDZJt6ojfSg9iVz_KfnG?usp=sharing
this is ,pynb file

Jaswanth-Devarinti, 2021-03-20:
Did you get it?

paulvangentcom, 2021-03-20:
Thanks, will have a look over the weekend.

-Paul

Jaswanth-Devarinti, 2021-03-20:
Ok, thanks.
Actually we are using heartpy for a small mini project and cane across
this. If you can do it early, it'll be a lot helpful as we did all the work
till this part. 😅😅
Thank you, I'm Jaswanth

paulvangentcom, 2021-03-20:
Ah you're using that example smartwatch dataset. Be aware that the data is very intermittent (some sections contain data, some do not), see the whole plot here:

[image]

You would need to write some code to exclude sections where no signal is present, and only run the analysis on sections where there's a signal present (the exception you're getting is telling you there's no signal). There's an example of how you might approach this task at the end of [that same notebook](https://github.com/paulvangentcom/heartrate_analysis_python/blob/master/examples/3_smartwatch_data/Analysing_Smartwatch_Data.ipynb), but by no means is that the only approach possible.

Alternative is collecting some of your own data using a smartwatch or ppg sensor.

Cheers,
Paul

Jaswanth-Devarinti, 2021-03-20:
Thanks, I'll check it out.
😁😁

paulvangentcom, 2021-03-20:
Good luck, let me know if you need further assistance. Closing this for now.

Cheers, 
Paul

closed 2021-03-20 by paulvangentcom

Jaswanth-Devarinti, 2021-03-21:
Hello,
https://colab.research.google.com/drive/1OCnzXzQbcPkwE_AfCgGbpye0G7tgtfOw?usp=sharing
I gave editable axis and I did the same thing as you did in the last
function I am getting error as
"BadSignalWarning:
---------------- Could not determine best fit for given signal. Please
check the source signal. Probable causes: - detected heart rate falls
outside of bpmmin<->bpmmax constraints - no detectable heart rate present
in signal - very noisy signal (consider filtering and scaling) If you're
sure the signal contains heartrate data, consider filtering and/or scaling
first. ----------------"

Kindly solve it. Thank you.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

paulvangentcom, 2021-03-22:
It looks like you're still running the whole signal through heartpy, while large parts contain nothing. The exception is telling you what's going on: there's no (detectable) heart rate signal in the first portion passed to it. 

You still need to expand the example code from the notebook a bit to also _exclude_ the parts with no signal, right now it's only setting it to 0.

Alternatively you step over all exceptions and keep running the loop, those sections with sufficient heart rate signal present will then still return results.

Yet alternatively you can find plenty of free ppg datasets online.

Good luck!
Paul

Jaswanth-Devarinti, 2021-03-22:
Sorry, but I did try removing things, but it's not working.
Please help me out.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

paulvangentcom, 2021-03-22:
As I mentioned, you need to handle the exception somehow for the cases where no heart rate signal is present in the example data. If you pass a signal segment **without heart rate signal in it** heartpy will complain, and that's what the exception is telling you in detail.

You can use a try except loop for handling exceptions, [see the Python docs here for more info](https://docs.python.org/3/tutorial/errors.html)

Cheers,
Paul