paulvangentcom/heartrate_analysis_python #66: One bpm value with process_segmentwise function

talhaanwarch, 2021-04-13:
I am trying to apply `process_segmentwise` and expect to get more than one bpm for each segment. But I am getting only one
```
data=np.array([-0.00107312, -0.05913093, -0.01802106,  0.11138419,  0.21170328,
        0.23130038,  0.23967177,  0.39018603,  0.46939674,  0.1269173 ,
       -0.3261911 , -0.49816029, -0.39486021, -0.1786203 , -0.029952  ,
        0.04890505,  0.14561741,  0.40896461,  0.58242192,  0.21352672,
       -0.3140083 , -0.52905975, -0.44560881, -0.18818952,  0.08245881,
        0.2558152 ,  0.28758079,  0.35741451,  0.35442566, -0.02197558,
       -0.42447667, -0.50889029, -0.33338838, -0.03594535,  0.18757467,
        0.26776168,  0.26662546,  0.2967212 ,  0.26732199, -0.0231855 ,
       -0.3452031 , -0.43486596, -0.31861585, -0.11521908,  0.04640795,
        0.15845268,  0.22972329,  0.34827644,  0.35486368, -0.02732068,
       -0.42366786, -0.47741352, -0.24809378,  0.05365441,  0.15115142,
        0.11400313,  0.16729757,  0.40227564,  0.52019283,  0.1338924 ,
       -0.35744099, -0.5052905 , -0.39457881, -0.1833988 ,  0.01203365,
        0.14558485,  0.20251516,  0.28444683,  0.30570056,  0.02210506,
       -0.31700213, -0.38397508, -0.20787165,  0.00520782,  0.09960194,
        0.127685  ,  0.16178406,  0.30402201,  0.35056315,  0.01085095,
       -0.36301018, -0.45148872, -0.30424407, -0.02981254,  0.15841651,
        0.16890123,  0.13446538,  0.24412992,  0.34269749,  0.10057154,
       -0.24885807, -0.35687456, -0.27177761, -0.14270168, -0.051568  ,
        0.02645136,  0.14066088,  0.35999524,  0.45062664,  0.10903204,
       -0.29608082, -0.3902395 , -0.23352422, -0.02627405,  0.04847967,
        0.05629757,  0.10040585,  0.25117953,  0.35697366,  0.10030356,
       -0.30709505, -0.4871904 , -0.412141  , -0.20367286,  0.00668297,
        0.17543259,  0.31126852,  0.53357413,  0.60526009,  0.15062919,
       -0.3815298 , -0.54414151, -0.40480428, -0.17383467, -0.03124084,
        0.09710991,  0.24403656,  0.44168115,  0.49973359,  0.10516081,
       -0.37039101, -0.51802543, -0.37590009, -0.11398634,  0.0893412 ,
        0.22551729,  0.29389534,  0.33862643,  0.28306611, -0.04489183,
       -0.37890703, -0.45384924, -0.29831726, -0.05574599,  0.09292824,
        0.14206961,  0.1735937 ,  0.32813233,  0.40836557,  0.07293123,
       -0.34739369, -0.47818101, -0.35569512, -0.12636987,  0.04429917,
        0.15025109,  0.24121963,  0.4182992 ,  0.49747832,  0.12973301,
       -0.32499562, -0.47813986, -0.38131286, -0.1900178 , -0.06494444,
        0.06058535,  0.23190519,  0.45549751,  0.5139708 ,  0.12071927,
       -0.34198754, -0.48442775, -0.36383722, -0.13255895,  0.0512353 ,
        0.17001333,  0.25314063,  0.36566019,  0.3684586 ,  0.02070558,
       -0.37133459, -0.47426545, -0.31890111, -0.05738652,  0.09485886,
        0.11301817,  0.14074772,  0.32165961,  0.41936488,  0.08938162,
       -0.31710168, -0.41762081, -0.28594246, -0.08291651,  0.08788089,
        0.21618437,  0.2552013 ,  0.26565618,  0.2112068 , -0.05200989,
       -0.29993416, -0.3506981 , -0.25768327, -0.10731679, -0.01344311,
        0.03176568,  0.1231765 ,  0.37906194,  0.49794883,  0.12290405,
       -0.31792482, -0.44742041, -0.3257845 , -0.06300943,  0.15410606,
        0.21973277,  0.195107  ,  0.26652339,  0.29898692,  0.00115041,
       -0.35181988, -0.45009013, -0.3221529 , -0.08729997,  0.08787529,
        0.14899414,  0.16998147,  0.33667452,  0.45127475,  0.10922899,
       -0.35352669, -0.50792013, -0.38207707, -0.1432262 ,  0.03834687,
        0.12420904,  0.16985284,  0.3976458 ,  0.57798634,  0.20917472,
       -0.33220752, -0.52960625, -0.36514452, -0.05957937,  0.09263803,
        0.10628904,  0.1486931 ,  0.31841991,  0.39884253,  0.09573288,
       -0.29118707, -0.41223978, -0.30407935, -0.11872425,  0.0228578 ,
        0.1329284 ,  0.19489777,  0.22653787,  0.20136718, -0.0223617 ,
       -0.27712476, -0.34236042, -0.22287744, -0.02944701,  0.0988922 ,
        0.16053124,  0.19743804,  0.30258611,  0.33707336,  0.03112066,
       -0.33533412, -0.44788962, -0.34643482, -0.12372072,  0.08591725,
        0.20535329,  0.24283919,  0.33832518,  0.3752323 ,  0.04190256,
       -0.35358746, -0.44515306, -0.27361462, -0.01151028,  0.13850071,
        0.16312289,  0.1410875 ,  0.22996554,  0.30902742,  0.0534508 ,
       -0.28769137, -0.38152435, -0.26372406, -0.08548462,  0.00981736,
        0.02954015,  0.07820258,  0.31355083,  0.48736884,  0.19836479,
       -0.21660955, -0.35801762, -0.26678843, -0.11414953, -0.05970772,
       -0.05034538,  0.03608476,  0.27371579,  0.41650617,  0.12536925,
       -0.29819848, -0.46423438, -0.35295068, -0.09500911,  0.10569267,
        0.20486073,  0.28022152,  0.4481121 ,  0.4971009 ,  0.09709216,
       -0.37784578, -0.51876178, -0.3699033 , -0.10598156,  0.08605208,
        0.17307391,  0.20072568,  0.31348403,  0.35064895,  0.00412193,
       -0.39434633, -0.50295735, -0.35227029, -0.08863309,  0.07476593,
        0.13400469,  0.20320386,  0.41259314,  0.52107612,  0.13223231,
       -0.3673405 , -0.52457058, -0.37858303, -0.09823316,  0.09849802,
        0.15352479,  0.17267004,  0.33324942,  0.44385972,  0.13112098,
       -0.30518219, -0.46463623, -0.36432214, -0.16140969, -0.02606023,
        0.04259304,  0.13834605,  0.38952234,  0.55279586,  0.20747147,
       -0.29242888, -0.46889586, -0.34581968, -0.10406559,  0.07250676,
        0.14956426,  0.1757876 ,  0.2732293 ,  0.31421531,  0.01647504,
       -0.34540878, -0.44253829, -0.30290557, -0.06512188,  0.09724587,
        0.15679229,  0.19340217,  0.32696067,  0.38951492,  0.07864304,
       -0.31304409, -0.41594123, -0.26328708, -0.05782799,  0.03070842,
        0.06405026,  0.12452822,  0.30525747,  0.41350778,  0.10100725,
       -0.31382201, -0.44339292, -0.31781181, -0.07970475,  0.0850417 ,
        0.13379378,  0.16809736,  0.31684192,  0.36792974,  0.04354825,
       -0.31110315, -0.38016044, -0.23701201, -0.03336602,  0.08243483,
        0.12634757,  0.1495003 ,  0.22211338,  0.22881385, -0.04012741,
       -0.31713079, -0.33630681, -0.16892347,  0.04050034,  0.18512663,
        0.28849841,  0.39685711,  0.57790848,  0.57962947,  0.00802581])    
```
    
    
`_,res=hp.process_segmentwise(data,sample_rate=14)
`

paulvangentcom, 2021-04-14:
The default segment width used is 120 seconds. This is larger than the 30 seconds of your provided data stream, which is why you only get data for one segment back.

[You can find the documentation of the specific function here](https://python-heart-rate-analysis-toolkit.readthedocs.io/en/latest/heartpy.heartpy.html#heartpy.process_segmentwise)

-Paul

talhaanwarch, 2021-04-14:
Thanks, Paul. I thought its 120 data points instead of seconds.

closed 2021-04-14 by talhaanwarch

paulvangentcom, 2021-04-14:
No worries, glad it helped!