paulvangentcom/heartrate_analysis_python #70: PPG general question

arashtavakoli, 2021-05-05:
Hi,

Thank you so much for this awesome package. I have been using it and unfortunately, it often misses lots of the peaks in my own PPG data. I have used the code below on my data and I have also attached an example of how it labels almost all peaks as rejected. Any suggestion would be much appreciated. 

[image]

df_final["PPG1_clean"] = hp.filter_signal(df_final["scaled_ppg"], [0.7, 2.5], sample_rate=90, 
                            order=3, filtertype='bandpass')
for s in [[0, 10000], [20000, 30000], [30000, 40000], [40000, 50000]]:
    wd, m = hp.process(df_final["PPG1_clean"].values[s[0]:s[1]], sample_rate = 90, 
                       high_precision=True, clean_rr=True)
    hp.plotter(wd, m, title = 'zoomed in section', figsize=(12,6))
    peaks = nk.ppg_findpeaks(df_final["PPG1_clean"].values[s[0]:s[1]],show=True)
    hp.plot_poincare(wd, m)
    plt.show()
    #hrv_indices = nk.hrv(peaks["PPG_Peaks"], sampling_rate=240, show=False)
    #print(hrv_indices["HRV_RMSSD"])
    for measure in m.keys():
        print('%s: %f' %(measure, m[measure]))

paulvangentcom, 2021-05-17:
Hi Arash,

It seems there is substantial noise still in the signal. Can you share a CSV file with me?

Paul

arashtavakoli, 2021-05-17:
Hi Paul,

Thank you so much for your reply. Here is a csv uncleaned file. This file is resampled at 100hz. I am also happy to meet you online if you have any suggestions. Thank you so much for your help. 

https://drive.google.com/file/d/1imQk9Rc4MyoAFeODQaCTBHWyWaJ3Nz8E/view?usp=sharing

Best,
Arash

paulvangentcom, 2021-06-29:
Hi Arash,

I had a quick look, find a quick notebook with some analysis here: [Arash.zip](https://github.com/paulvangentcom/heartrate_analysis_python/files/6731968/Arash.zip)

The signal is really quite noisy, and segments will not give a good solution.

Paul

closed 2021-06-29 by paulvangentcom

arashtavakoli, 2021-07-03:
Hi Paul,

I took my time to really understand what you did in the notebook. Thank you
soo much. I am just wondering how do you find the values for the bandpass
filter? basically, this number can change depending on how much movement
there is I guess. Also, do you have any general suggestions in this regard?
I understand that you use the Poincare plot but I was not sure if there is
anyway to actually automate this process? Also as you mentioned it is way
too noisy. Do you think such signal is of any use or I should just
completely throw it away?

Sorry for too many questions. I would be very happy to also talk to you on
phone.

Cheers,
Arash

paulvangentcom, 2021-07-03:
Hi Arash,

The bandpass values I pick based on expectation of the bpm range. 0.7 - 2.0 Hz in this case would translate to 42 - 180 bpm. You can expand the range if you expect higher values. Lower I wouldn't quickly expect.

The noisiness is in the original signal, I did include the poincare plot but that was for illustrative purposes mainly : ). If possible I would definitely consider getting another way of collecting the signal. pulsesensor.com has quite good ones you can put on fingertips and earlobes. Quite a lot of peaks are discarded by heartpy, most likely because the underlying signal is too influenced by the present noise, which shifts the peaks to unexpected and unlikely positions. 

You can use this type of signal quite well for BPM only (it's resistant to noise) but not really for the HRV metrics (very sensitive to noise)

Cheers
Paul