paulvangentcom/heartrate_analysis_python #71: heartpy for RPPG-HRV wrong values

MAFEMV, 2021-05-14:
Hello I've been using your repository for RPPG, to extract vitals signs from a blod volumne pulse signal obtain from short videos (10-20s, (25fpm)), I have been trying your repo for this task due to you have used for PPG signals. In the per-processing stage I filtered and normalized the BVP signal. Even though I got normal Heart rate and Breathing rate values, the HRV values such as RMSSD and SDNN are far away from its normal values.
In your opinion, what could be the cause to obtain the wrong HRV values? ... noisy signal? wrong counting of peaks? to short videos/signal?
Thank you in advance.
Maria Montoya

paulvangentcom, 2021-05-17:
Hi Maria,

Generally I've found rPPG isn't very suitable for HRV metrics. It is fine to get a BPM from it, since that is a metric that is quite robust against outliers. However, HRV metrics are very sensitive to outliers, and quickly get distorted with low quality signals like rPPG.

In the past I've had a few master students working on this and not really finding a suitable solution in the process.

Cheers,
Paul

MAFEMV, 2021-05-18:
Hi professor Paul,
Thank you for your response, we will keep this information in mind.
Regards

closed 2021-05-18 by paulvangentcom