paulvangentcom/heartrate_analysis_python #72: Feature request : Adaptive filters

veeresh-dammur, 2021-05-17:
Hi, 

Thanks for the wonderful Python library! 
I am using [AFE4404 ](https://www.ti.com/product/AFE4404)+ [SFH7060](https://www.osram.com/ecat/BIOFY%C2%AE%20SFH%207060/com/en/class_pim_web_catalog_103489/prd_pim_device_2220014/) to record the raw PPG signals.  I am using it to determine the HRV metrics. But whenever there is a movement with respect to hands, then I am missing heartbeat peaks. My processing pipeline is shown below,  

Raw data --> Moving average --> Band pass filter (0.4 to 5 Hz) --> hp.process (band_passed_signal, sampling_rate=50) --> RR interval 

1. Is there any way to improve the signal quality so that we get still better HRV metrics. 
2. It would be nice if you could include the feature of Adaptive filters using acceleration data as reference signals.

paulvangentcom, 2021-06-29:
Thanks for the suggestion! I will have a look into adaptive filters. 

If you can share some example data with me I can have a look as well.

Pittconnect, 2021-08-20:
Hello @paulvangentcom this is a really good algo thanks for sharing. 
Here somebody has a data set that is from the watch sensor and the data has kind of odd format to it like slashes https://forum.pine64.org/showthread.php?tid=14243&pid=99003#pid99003
Daniel says writing your own logger to create a hrv.data may be the first step in the right format. So it can be analyized I quite do not understand the data.log it has 3 digit number Heart rate is in two digit so to me it seems this may be frequency reading like 530hz right ? But is it 24hz frequency it can be converted to bpm https://www.convertworld.com/en/frequency/hertz/hz-to-bpm.html