paulvangentcom/heartrate_analysis_python #76: Speed up heartpy

alexxony, 2021-06-28:
my ecg dataset is 48hour. so It will take so long time. 
can you give an advice of speed up?

paulvangentcom, 2021-06-28:
Hi Alexxony,

Processing shouldn't take more than a few minutes, what is the sample rate and what flags do you use?

Paul

alexxony, 2021-06-29:
I used my data; it is 48 hours ecg which includes extreme motion
artifacts.  i runned it on colab. it tooks over 10 hours.

+
enhance_ecg_peaks make omit of data?I applied it on data(len is 50000),
applied data was 49991.

paulvangentcom, 2021-06-29:
That is definitely atypical, it should be a lot faster. Can you share the data with me so I can do some testing here?

Enhance peaks does not discard any data, it's a simple multiplication and scaling operation.

Paul

paulvangentcom, 2021-07-20:
Closing due to inactivity, feel free to reopen if you still need assistance

closed 2021-07-20 by paulvangentcom