paulvangentcom/heartrate_analysis_python #89: What "s" stand for?

BaggioMarco, 2021-12-15:
Hi,

thank you so much for this great pack!
I'm just wondering which is the meaning of "s" that is listed among the other indices.
It get printed by this block below:

```
#display measures computed
for measure in m.keys():
    print('%s: %f' %(measure, m[measure]))
```

thanks in advance

paulvangentcom, 2021-12-16:
It's a component of the poincare analysis. 

Consider this snippet of code:

```python
import heartpy as hp

data, _ = hp.load_exampledata(2)
wd, m = hp.process(data, sample_rate = 100.0, high_precision = True)
hp.plot_poincare(wd, m)
```

This gives the following plot:
[image]

This plots the peak-peak intervals with RR(n) on the x-axis and RR(n+1) on the Y axis.

SD1 is the standard deviation of intervals along the identity line (x=y)
SD2 is the standard deviation of intervals perpendicular to the identity line
S is the *surface area* of the ellipsoid described by SD1 and SD2

Hope this answers your question :)

Cheers

paulvangentcom, 2021-12-16:
Closing for now. Feel free to re-open if you have additional questions or comments

closed 2021-12-16 by paulvangentcom