paulvangentcom/heartrate_analysis_python #90: customize frequency bands for frequency-based analysis?

KathMaki, 2021-12-30:
Thanks for a great program. I am currently analyzing rat data where HRs are much higher vs. human (~300-450 bpm at times). In the literature, frequency bands are different for rodents as compared to humans.  I have seen LF ranges around 0.2-0.8 Hz and 0.15-1.5 Hz, HF ranges have been reported as 0.8-2.8 Hz and 1.5-5 Hz 

Is it possible to add a dictionary or flag to customize the frequency bands so I can use the freq or process functions for my HRV analysis? 

Additionally, I have ~24 hour recordings for each intervention day (I use aortic BP but the peak-peak detection has been working well in your program). 

My thought is to first segment out the signal files by hour (i.e. so only 8am values are calculated then 9am etc.) and then use a sliding window of 3-4 minutes (as heart rates are higher so more beats are obtained) using "process_segmentwise" to calculate the HRV parameters. My goal is to have one mean HRV value (for my metrics of interest) for each hour, per animal. Does this seem appropriate? Thanks in advance!

james-sexton96, 2022-01-03:
@KathMaki Although there isn't a specific way to modify the VLF, LF, and HF right now through the hp functions (although you could fork your own source code), I think your best bet is to use the raw 'psd' and 'frq' information stored in the working data output of the process function. 

If you want FFT output you would specify the following options in your process or process_segmentwise argument:

 `
calc_freq = True, freq_method = 'fft'
`

You would essentially be saving two lists containing the raw FFT data for each segment. These then could be used to calculate your own LF or HF bands according to the murine HRV literature :) 

See the following source code I've slightly modified as an example:

`
df = frq[1] - frq[0]
Murine_LF = np.trapz(abs(psd[(frq >= 0.2) & (frq < 0.8)]), dx=df)
`

Based on what you are saying it is also a good idea to modify the BPM min and max arguments so that no erroneous peak rejection happens to your data (although you've probably already sorted this out). 
Additionally, I think it makes sense that you would use process segmentwise to split up the data to gather metrics over time and look at temporal trends.
However, I'd suggest longer segments for frequency based metrics. These really need about 5 - 10 minutes of very, very clean data to be reliable enough to look at temporal trends. 
Regardless of the shift in LF and HF bands which may allow you to get away with smaller windows, more data is always better for these calculations and you have a lot to work with given the 24 hour recording. 

Happy hunting!

KathMaki, 2022-01-03:
Thank you for the response and help with the source code!

To be clear, are you saying for process_segmentwise if I wanted to process each hour at a time and use 5 min segments with 50% overlap I would use this code (after subsetting the data by hour):
wd, m = hp.process_segmentwise(signal_data, sample_rate = 250.0, segment_width = 300, segment_overlap = 0.5, segment_min_size=300, bpmmin=150, bpmmax=400, calc_freq=True, freq_method='fft') #segment min size would be to throw away any segment that is under 300 sec....bpm min and max are the expected ranges for rat HRs. I read standard process arguments can also be used in process_segmentwise which is important in my case since the HR ranges are needed so peaks are correctly detected.

And if I wanted to use 10 min segments I would use this:
wd, m = hp.process_segmentwise(signal_data, sample_rate = 250.0, segment_width = 600, segment_overlap = 0.5, segment_min_size=600, bpmmin=150, bpmmax=400, calc_freq=True, freq_method='fft')

Do you recommend replacing outliers as a default? My signal is clean and so far there haven't been issues with peak detection but curious what you think about this.

In regards to the FFT portion:
 I see that you said with the raw fft data I would be saving two lists. When examining the output of the process function I see these two arrays returned (frq and psd). Can you expand on what is returned here for my understanding? If you have any resources to point me to as well that would be super helpful as I don't want to take too much of your time. 

Finally, I am used to seeing LF and HF power reported in ms^2 in the literature. I see with the calc_fd_measures, there is a square_spectrum option that would give me the output I am used to seeing (I believe). But if I am going to be accessing the raw fft data to calculate my own frequency bands based on the rat literature, would it be best to get the data in ms and then square the results at the end or would it be cleaned to get the output squared from the beginning since that is how I usually see it reported? 

Again, thank you so much!