paulvangentcom/heartrate_analysis_python #91: PPG input file or array ?

saranzeb, 2022-01-02:
i am really looking for some simple  answers
1) what should be the form of our input data (arrays) which 
we are going upload to hp .load  or hp.get_data. either its shoud be the 
BVP values with timing information ?  the instananoes time  of the bvp readings 
or the total time of BVP reading,i.e the total timing of video or window size ?,because from  PPG signal we can have only 
these two  Readings (information)  BVP values and time right? 
they shoud be both in same file like two arrays of BVP and time 
2) secondly i am not getting the sampling rate here it will be depending on what 
how we will choose it ?

paulvangentcom, 2022-01-03:
Hi Saran,

1. The form of the input data should be BVP values. When processing the data with heartpy.process() you specify the sample_rate which encompasses the timing information. If you do now know the samplerate but have a timing array, you can compute that with the [get_samplerate_mstimer()](https://python-heart-rate-analysis-toolkit.readthedocs.io/en/latest/heartpy.heartpy.html#heartpy.get_samplerate_mstimer) or [get_samplerate_datetime()](https://python-heart-rate-analysis-toolkit.readthedocs.io/en/latest/heartpy.heartpy.html#heartpy.get_samplerate_mstimer) methods.

2. The sample rate is very important to specify correctly, as it reflects how the heart rate is spaced over time. Without this value the analysis is not possible. Refer to the above methods if you have timer data but no sample rate, for methods that can help you compute it.

-Paul

JesseShaw, 2022-01-03:
If helpful, I can supply a BVP file and code to extract UTC and sample rate for Empatica devices.

saranzeb, 2022-01-04:
Thank you sir for  detailed answer it will be great if you can provide some files (BVP and timing ) with code so I can try them first and then can relate to my files, as i will be able to check the values and format .
in my case I am trying  to give Just the BVP file and a random sampling  rate, to the heart.py process (BVP) as I have just the BVP file  and want  just have a try but its gives me this error
 
''Could not determine best fit for given signal. Please check the source signal.
 Probable causes:
- detected heart rate falls outside of bpmmin<->bpmmax constraints
- no detectable heart rate present in signal
- very noisy signal (consider filtering and scaling)
If you're sure the signal contains heartrate data, consider filtering and/or scaling first''

[image]

my BVP file is a reading of just BVP values and its just an array of 2084 values saved in excel file which is the BVP values of a sample.  Along with  it i also have a video of 41 seconds, and a heart rate of 37values taken per second for the same sample , as i am calculating the heart rate in (bpm) from video so  for that I also need the heart rate from these  BVP values which i can compare with heart rate from video and for video i have to calculate the heart rate in bmp too  in the same way.

saranzeb, 2022-01-04:
Thank you sir for detailed answer it will be great if you can provide some
files (BVP and timing ) with code so I can try them first and then can
relate to my files, as i will be able to check the values and format .
in my case I am trying to give Just the BVP file and a random sampling
rate, to the heart.py process (BVP) as I have just the BVP file and want
just have a try but its gives me this error

''Could not determine best fit for given signal. Please check the source
signal.
Probable causes:

   - detected heart rate falls outside of bpmmin<->bpmmax constraints
   - no detectable heart rate present in signal
   - very noisy signal (consider filtering and scaling)
   If you're sure the signal contains heartrate data, consider filtering
   and/or scaling first''
   - i also have attached the screenshot below

my BVP file is a reading of just BVP values from **VIPL_HR ** dataset and
its just an array of 2084 values saved in excel file which is the BVP
values of a sample. Along with it i also have a video of 41 seconds, and a
heart rate of 37values taken per second for the same sample , as i am
calculating the heart rate in (bpm) from video so for that I also need the
heart rate from these BVP values which i can compare with heart rate from
video and for video i have to calculate the heart rate in bmp too in the
same way.

paulvangentcom, 2023-03-31:
There's example files included, you can use the method load_exampledata() to load these

See the [quickstart here](https://python-heart-rate-analysis-toolkit.readthedocs.io/en/latest/quickstart.html)

Or the [api reference for the method here](https://python-heart-rate-analysis-toolkit.readthedocs.io/en/latest/heartpy.heartpy.html#heartpy.load_exampledata)

closed 2023-03-31 by paulvangentcom