paulvangentcom/heartrate_analysis_python #102: Why cutoff=0.05Hz in remove_baseline_wander()?

yjhan99, 2023-01-17:
Hi.
Although it is not a code-related issue, but I leave a question since it's hard to find related information.

I wonder **why a notch filter of cutoff frequency 0.05Hz** is used for remove_baseline_wander function in hp.filtering.

The document says it is a function to remove baseline wander from ECG signals,
and according to related studies, _Baseline wander is a low-frequency noise of around 0.5 to 0.6 Hz. To remove it, a high-pass filter of cutoff frequency 0.5 to 0.6 Hz can be used._
But I found that you recommend a notch filter of cutoff frequency 0.05Hz, and it works well with noisy ECG signal. In contrast, cutoff frequency 0.5 or high-pass filter does not work well...

I wonder why this is happening. It would be great if you could answer my question.
Thanks in advance!

paulvangentcom, 2023-03-31:
Hi there,

The quote you give: `according to related studies, Baseline wander is a low-frequency noise of around 0.5 to 0.6 Hz. To remove it, a high-pass filter of cutoff frequency 0.5 to 0.6 Hz can be used.` is about `high-pass filters`, which are a different thing from `notch filters`. 

`High-pass filters` can be used to attenuate unwanted low frequency content from the signal (not just the baseline wander). You can view high-pass filters as something that will reduce the amplitude of all frequencies below a given threshold. Say you set it fo 0.5Hz, then the lower relative to 0.5Hz a frequency component is, the more it will be suppressed.

The main difference with a `notch filter` is that the notch filter will suppress a specific small frequency range, rather than "everything below" a threshold (or "everything above" in case of a low-pass).

This image might clarify a few things intuitively: 
[image]

And there's some more info where I took the image from: https://www.vectornav.com/resources/inertial-navigation-primer/math-fundamentals/math-filtering

paulvangentcom, 2023-03-31:
Closing for now, feel free to re-open or start a new issue if you still have questions

closed 2023-03-31 by paulvangentcom