paulvangentcom/heartrate_analysis_python #103: Peak detection fails for high DC levels

josueportiz, 2023-03-20:
In `fit_peaks()`, the moving average values to test list (`ma_perc_list`) starts at 5 %. Signals with high DC values related to the AC component (e.g. some PPG signals) add a high value to test so that the starting 5% leads to moving average values higher than any of the peaks of the signal. 

Please, consider add fine grained values to the `ma_perc_list`: 0, 1, 2, 3, 4 %. Otherwise it does not work.

paulvangentcom, 2023-03-31:
Thanks for the suggestion, I will put it on the list for the next version. Currently picking up heartpy again.

SahebehDadboud, 2023-04-04:
@paulvangentcom  that's one of the reasons I can't use Hearpy package as a built-in library, so I had to clone the project. If you add `ma_perc_list` as a parameter to the function (which I did already), it would be much more beneficial. Currently I had to divide the original  `ma_perc_list` by 10 to let rolling mean find the peaks in our device.