paulvangentcom/heartrate_analysis_python #113: Missing dependency when importing heartpy

glopesdev, 2025-02-13:
The below import creates a missing dependency from `pkg_resources` which is not declared in the `heartpy` package itself.

https://github.com/paulvangentcom/heartrate_analysis_python/blob/e3d80c2eff1967120cf5210b1f88cb23347246b0/heartpy/datautils.py#L6

The below is enough to reproduce:

```
pip install heartpy
python -c "import heartpy"
```

This will throw the following stacktrace:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\dev\python\heartpy\.venv\lib\site-packages\heartpy\__init__.py", line 1, in <module>
    from .heartpy import *
  File "C:\dev\python\heartpy\.venv\lib\site-packages\heartpy\heartpy.py", line 15, in <module>
    from .datautils import get_data, get_samplerate_mstimer, get_samplerate_datetime,\
  File "C:\dev\python\heartpy\.venv\lib\site-packages\heartpy\datautils.py", line 6, in <module>
    from pkg_resources import resource_filename
ModuleNotFoundError: No module named 'pkg_resources'
```

In the documentation it is stated:

https://github.com/paulvangentcom/heartrate_analysis_python/blob/e3d80c2eff1967120cf5210b1f88cb23347246b0/heartpy/datautils.py#L57-L58

Sadly, the above import makes this statement incorrect, as `pkg_resources` is imported no matter what at the top of the file.

paulvangentcom, 2025-12-29:
Thanks for raising this, and indeed this is an issue. I have a fix in mind using the now more pythonic `importlib` which I will push tonight or tomorrow.

Heartpy is due for a major facelift soon, too.

paulvangentcom, 2025-12-30:
a fix is live in the repository, please uninstall your current heartpy `pip uninstall heartpy`, pull the latest from git and reinstall from the repository folder `python setup.py install`

Unfortunately I am locked out of pypi for the moment. I have raised a ticket to have this fixed, and will update the version available through pip as soon as possible

closed 2025-12-30 by paulvangentcom